jlas1/Klicky-Probe #266: Wrong attachmove formula?

dbarwikowski, 2024-09-03:
Hey klicky gods, 

Ive installed klipper on my old Anet A8 and it was fun project.
Now im adding klicky/euclid probe to it. Currently fighting with some macros and design problems but I'm getting closer every day 😁

I would love to get some help with klicky macros.
First of all i have problem with attachmove.
Dock is next to z1 motor. Mounted to the motor at - 12, 0, -1. 
To attach probe, carriage must home X and go down slowly until probe is touched.
I've figured it out. But to move out of the dock carriage must travel right, 32mm on X axis.

Attachmove is 32, 0, 0 but when macro tries to do it, it goes the wrong way.
In klicky macro I see that move is calculated like this
Dock_position - attachmove.
And it works for everybody... Except me 😁

Please help

jlas1, 2024-09-03:
Hello there, can you enable debug on klicky variables and paste the console output here, with your klippy.log.

That should help to clear things up.

dbarwikowski, 2024-09-04:
Your code to attach probe:

```
# Probe entry location
G0 X{docklocation_x|int - attachmove_x|int - attachmove2_x|int} Y{docklocation_y|int - attachmove_y|int - attachmove2_y} F{travel_feedrate}
# Attach probe
G0 X{docklocation_x|int - attachmove2_x|int} Y{docklocation_y|int - attachmove2_y} F{travel_feedrate}
G0 X{docklocation_x} Y{docklocation_y} F{dock_feedrate}
# Probe exit location
G0 X{docklocation_x|int - attachmove_x|int} Y{docklocation_y|int - attachmove_y|int} F{release_feedrate}
```

docklocation: -12, 0, -1
attachmove: 30, 0, 0

I will add console output later but I don't see how this could ever work for anyone

dbarwikowski, 2024-09-10:
@jlas1 do you need more info? :)