jlas1/Klicky-Probe #268: Please add macros for X-axis_twist_compensation

PhilBaz, 2024-10-17:
Please add macros for X-axis_twist_compensation

https://github.com/Klipper3d/klipper/pull/6624#issuecomment-2420026334

jlas1, 2024-10-17:
Hi, it seems that it's necessary to grab the probe before doing the AXIS_TWIST_COMPENSATION_CALIBRATE, can you confirm that's the case?

PhilBaz, 2024-10-18:
@jlas1 

I haven't tried that my self but I believe I did see Sineos say that on Klipper discourse...Ahh found it.

https://klipper.discourse.group/t/feature-request-dockable-probe-support-for-axis-twist-compensation/11744

Alia5, 2024-12-07:
> Hi, it seems that it's necessary to grab the probe before doing the AXIS_TWIST_COMPENSATION_CALIBRATE, can you confirm that's the case?

I tried to setup axis-twist-compensation, however, the probe nor the z endstop stop the printhead from moving down.
I think there is more work to be done than just attaching the probe...