jlas1/Klicky-Probe #273: Axis homing erratic

pmastren, 2024-12-16:
Installed Klick=Probe on custom built corexy machine. When homing XY, it would home X then skip homing Y.  Changing variable_override_homing to 'Y" would then home Y but skip homing X.  The solution is to add M400 to the beginning of each homing segment to wait for the previous motion to complete.

janderholm, 2025-04-05:
This looks like a duplicate a #256.

As you say, a M400, but preferably at the end of each homing segment, seems to do it.

jlas1, 2025-04-06:
Remember that klicky by itself does not use _home_x or _home_y, it just uses it when it finds the macros.
Whith that being said, i see no issue in adding a M400 after G28 x,y and Z

closed 2025-04-06 by jlas1