jlas1/Klicky-Probe #275: troubleshooting probe attachment/detachment

tibers, 2025-01-19:
Hey folks

I have the configs all working and I am at the part where we test the docking and undocking. I am getting errors like PROBE ALREADY ATTACHED.

I was hoping for a troubleshooting.md or similar readme file but I'm not seeing it. Any tips?

Thanks!

jlas1, 2025-01-20:
do a query_probe command.
You should get triggered when no probe attached
You should get open when the probe is attached at the toolhead
You should get triggered when the probe is attached at the toolhead and the probe pin is pressed

If it's reversed, add a ! before the probe pin definition, if it's allways open, or triggered check the cabling and if the pin on the mcu is a endstop pin and is correcly configured

tibers, 2025-01-20:
Thanks for the tips!

I actually messed around with this way too much last night. 

[image]

Apparently `proximity` isn't just a label on the EBB SB2240 and 2209. It does something else funky. I also tried `^` notation for the signal pin and that didn't work either. I moved it to the four pin endstop for `PB5` and that fixed it and it worked. It's something funky in the board, not your macros. Thanks for this project and answering this issue. :)

closed 2025-01-20 by tibers

jlas1, 2025-01-24:
thanks for the feedback, glad everything is working now.