jlas1/Klicky-Probe #279: dock_on_zhome state confusion

bombela, 2025-02-28:
If  `dock_on_zhome` is true (the default), the probe is docked before Z home but not after.
If false, this reverses. The probe is not docked before Z home, but is docked after.

In other words, it is impossible to Z home without the probe being docked at some point.

This strikes me as a bug in the logic. This should either be two config variables and/or a dynamic parameter for optionally docking post Z homing.

Because in the current state, one cannot optimize a sequence of home XYZ, QGL, auto-z-cal, without the probe being docked and re-attached multiple times. I have hacked the macros quick and dirty for my application, and I figured I should bring this to your attention.

https://github.com/jlas1/Klicky-Probe/blob/ab86f91d47ac7353a0a7c7e0de26542dca00f63e/Klipper_macros/klicky-macros.cfg#L781-L801

Koza666, 2025-03-09:
Faced the similar issue with slighty different setup (triple z printer and z-tilt routine). For me the workaround was to home X and Y axis separetly and add proble lock before homing Z, to avoid docking probe multiple times. My start print macro looks something like this:

```
[gcode_macro START_PRINT]
gcode:
    {% set BED_TEMP = params.BED_TEMP|default(60)|float %}
    {% set EXTRUDER_TEMP = params.EXTRUDER_TEMP|default(190)|float %}
    # Start bed heating
    M190 S{BED_TEMP}
    BED_MESH_CLEAR

    # Use absolute coordinates
    G90

    # Reset the G-Code Z offset
    SET_GCODE_OFFSET Z=0.0
    
    # Home the printer
    G28 X Y
    Attach_Probe_Lock
    G28 Z
    Z_TILT_ADJUST
    BED_MESH_CALIBRATE
    Dock_Probe_Unlock
    Smart_Park
    M109 S{EXTRUDER_TEMP}
    LINE_PURGE

```

jlas1, 2025-04-06:
@bombela is the correct solution in the current state of klicky macros

closed 2025-04-06 by jlas1

bombela, 2025-04-06:
I am not sure to understand the question.

Do you mean you have pushed some changes and want me to take a look?

jlas1, 2025-04-07:
sorry for not being clearer, what i wanted to say, is that this approach is the best one to minimize probe docking and attaching:
    # Home the printer
    G28 X Y
    Attach_Probe_Lock
    G28 Z
    Z_TILT_ADJUST or QGL
    BED_MESH_CALIBRATE
    Dock_Probe_Unlock