jlas1/Klicky-Probe #284: D2F-5 vs  D2F-01

janderholm, 2025-04-05:
According to the datasheet for the D2F series of microswitches, the D2F-01 seems like the better choice and D2F-5 the wrong choice in most cases.

The D2F-01 uses gold alloy contacts and has a minimum applicable load of 1mA at 5V. The D2F-5 uses silver (which oxidizes with time) and a minimum load of 100mA.

Given that most microcontrollers run at 3.3V and has internal pull-ups of 30K - 100K or so (Resulting in a current of ~0.1mA - 0.03mA), the D2F-5 seems like a very poor choice compared to the D2F-01. In fact to run the D2F-01 properly in spec a 4K7 resistor to 5V is needed. Although I'm sure it works well enough in most cases on the internal pull-ups.

It seems to me that choosing the D2F-5 has a larger likelyhood of being unreliable after some time.

I may be missing some common machine that has large currents passing through the switch, but otherwise perhaps the recommendation should be changed to D2F-01?

Datasheet (Page 2 - Contact Specifications, Page 5 - Using Micro Loads):
https://omronfs.omron.com/en_US/ecb/products/pdf/en-d2f.pdf

jlas1, 2025-04-06:
Actually, looking at the data that we collected when testing unklicky, all are very similar, the one that got the best results was actually D2F-L.

[image]

all are fit for service through.

janderholm, 2025-04-06:
The issue with the D2F-5 isn't so much about the accuracy, but that the silver oxidizes with time. At higher currents the oxidization removes itself, but with only low currents it becomes a barrier.

It will probably require some sort of accelerated aging to test though which is difficult and something the manufacturers have done already. 

Gold doesn't oxidize but I think higher currents destroy any gold plating.

There's a little bit about the topic here:
https://www.digikey.com/en/blog/benefits-of-gold-plated-vs-silver-contacts

jlas1, 2025-04-07:
i understand your point, but i dont think it's the only issue we face on our use case, it's very different from the original propose, the switch always trigger when they fail the accuracy tests, but are just not accurate enought for our use cases.
For our use case, it's all alout the accuracy.

One of the reasons pcb klicky was developped was to use a resistor to put the switch in the correct current, but afaik, there has been no noticiable increase in switch life or accuracy.